PX4/PX4-windows-toolchain #6: need to install pyymal manually after installation

cycle95, 2018-12-22:
following the Installation Instructions of Windows Cygwin Toolchain, when building the code using " make px4_sitl jmavsim" for the first time, it returns a error says failed to import yaml and I have to install pyyaml mannually to make it work.

OS: Windows 10 Pro
Error log:
[19/698] Generating serial_params.c                                                                                     FAILED: generated_params/serial_params.c                                                                                cd /cygdrive/d/PX4_ToolChain/home/Firmware/build/px4_sitl_default/src/lib/parameters && /usr/bin/cmake.exe -E make_directory /cygdrive/d/PX4_ToolChain/home/Firmware/build/px4_sitl_default/generated_params && /usr/bin/python /cygdrive/d/PX4_ToolChain/home/Firmware/Tools/serial/generate_config.py --params-file /cygdrive/d/PX4_ToolChain/home/Firmware/build/px4_sitl_default/generated_params/serial_params.c --serial-ports --config-files /cygdrive/d/PX4_ToolChain/home/Firmware/src/drivers/distance_sensor/sf0x/module.yaml /cygdrive/d/PX4_ToolChain/home/Firmware/src/drivers/distance_sensor/tfmini/module.yaml /cygdrive/d/PX4_ToolChain/home/Firmware/src/drivers/distance_sensor/ulanding/module.yaml /cygdrive/d/PX4_ToolChain/home/Firmware/src/drivers/distance_sensor/leddar_one/module.yaml /cygdrive/d/PX4_ToolChain/home/Firmware/src/drivers/gps/module.yaml /cygdrive/d/PX4_ToolChain/home/Firmware/src/modules/mavlink/module.yaml                                 Failed to import yaml.                                                                                                  You may need to install it with 'sudo pip install pyyaml'                                                                                                                                                                                       Traceback (most recent call last):                                                                                        File "/cygdrive/d/PX4_ToolChain/home/Firmware/Tools/serial/generate_config.py", line 13, in <module>                      import yaml                                                                                                         ImportError: No module named yaml                                                                                       [24/698] Generating git version header                                                                                  ninja: build stopped: subcommand failed.                                                                                make: *** [Makefile:182：px4_sitl] ERROR 1

cycle95, 2018-12-22:
Sorry, just figured it out that it might be my network connecting issue

bys1123, 2018-12-22:
Thanks you open a issue when you translate this doc on crowdin.
I think it's not a network connecting issue, msi file contains all packages.
Let's find this real problem together.

closed 2018-12-23 by cycle95

bys1123, 2018-12-23:
Looks like you are using 0.3 version, use latest v0.5 msi will fix.