PX4/PX4-windows-toolchain #7: Is it possible to add GDB debugger in this toolchain?

oneWayOut, 2019-01-12:
The windows toolchain is awesome.   I tried to debug the program in simulation but didn't find gdb. Could you add GDB in this toolchain?

bys1123, 2019-01-12:
PX4/Firmware#10911

hamishwillee, 2019-01-13:
A question for you @MaEtUgR !

oneWayOut, 2019-01-13:
Thank you! 谢谢！　@bys1123

MaEtUgR, 2019-01-13:
@oneWayOut Yes it's theoretically well possible. I think it should even work out of the box if you use gdb which is shipped with cygwin. I can give you some hints but I guess won't have the time to add it myself anytime soon.
- Download the cygwin installer https://cygwin.com/setup-x86_64.exe
- start it and choose C:\PX4\toolchain\cygwin64 as installation folder
- search for the packages you need e.g. gdb
- continue and install them
- run them from the `run-console.bat` terminal

If you figure it out and get the debugger working I'm happy to incorporate pull requests e.g. extending this script: https://github.com/PX4/windows-toolchain/blob/master/toolchain/cygwin64/install-cygwin-px4.bat

oneWayOut, 2019-01-14:
@MaEtUgR Thanks for your prompt response. 
Following your instructions, I installed gdb 7.9.1-1 successfully. 
But the command " make px4_sitl_default jmavsim___gdb " doesn't work properly. 
[image]
It seems that PX4.exe could not be found. But I checked that it  is definitely there.
 "make px4_sitl_default jmavsim" still works fine. 

Anyway, this toolchain is good enough for me, Thank you. If I need debug, I can do it in Ubuntu.

hamishwillee, 2019-01-14:
@MaEtUgR Would be good to understand why that toolchain doesn't find PX4 - we're so close to getting this working, would be good to get it all the way :-)

Any ideas?

MaEtUgR, 2019-01-15:
@hamishwillee You have a point, espacially because it should in principle work.

Now I took myself the time to try out the basic test, installed the gdb version currently recommended by cygwin `$ gdb --version` -> `GNU gdb (GDB) (Cygwin 8.0.1-1) 8.0.1`. Started the makefile with the previously mentioned command and got it running. The part which failed for @oneWayOut succeeded `Reading symbols from /cygdrive/c/PX4/home/Firmware/build/px4_sitl_default/bin/px4.exe...done.` and I got into the gdb console. I'm gdb terminal noob but after I found out I have to run the program first with the command `r`, I saw the application starting and some debug output what the threads are doing and got some memory address info of the stack frame and program counter. I'm not sure how to proceed from there but I'm sure if you know how to use gdb it is working fine.

Here's my console output:
```
MaEtUgR@Optiplex ~/Firmware
$ make px4_sitl_default jmavsim___gdb
ninja: Entering directory `/cygdrive/c/PX4/home/Firmware/build/px4_sitl_default'
[6/697] git submodule src/drivers/gps/devices
[17/697] git submodule src/lib/DriverFramework
[20/697] git submodule src/lib/ecl
[176/697] git submodule mavlink/include/mavlink/v2.0
[191/697] git submodule Tools/jMAVSim
[696/697] cd /cygdrive/c/PX4/home/Firmware/build/px4_sitl_default/tmp && /cygdrive/c/PX4/home/Firmware/Tools/sitl_run.sh /cygdrive/c/PX4/ho...ware/build/px4_sitl_default/bin/px4.exe gdb jmavsim none /cygdrive/c/PX4/home/Firmware /cygdrive/c/PX4/home/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /cygdrive/c/PX4/home/Firmware/build/px4_sitl_default/bin/px4.exe
debugger: gdb
program: jmavsim
model: none
src_path: /cygdrive/c/PX4/home/Firmware
build_path: /cygdrive/c/PX4/home/Firmware/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: "/cygdrive/c/PX4/home/Firmware/build/px4_sitl_default/bin/px4.exe" "/cygdrive/c/PX4/home/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/cygdrive/c/PX4/home/Firmware"/test_data
GNU gdb (GDB) (Cygwin 8.0.1-1) 8.0.1
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-pc-cygwin".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /cygdrive/c/PX4/home/Firmware/build/px4_sitl_default/bin/px4.exe...done.
warning: File "/cygdrive/c/PX4/home/Firmware/build/px4_sitl_default/tmp/rootfs/.gdbinit" auto-loading has been declined by your `auto-load safe-path' set to "$debugdir:$datadir/auto-load".
To enable execution of this file add
        add-auto-load-safe-path /cygdrive/c/PX4/home/Firmware/build/px4_sitl_default/tmp/rootfs/.gdbinit
line to your configuration file "/cygdrive/c/PX4/home/.gdbinit".
To completely disable this security protection add
        set auto-load safe-path /
line to your configuration file "/cygdrive/c/PX4/home/.gdbinit".
For more information about this security protection see the
"Auto-loading safe path" section in the GDB manual.  E.g., run from the shell:
        info "(gdb)Auto-loading safe path"
(gdb) Buildfile: C:\PX4\home\Firmware\Tools\jMAVSim\build.xml

make_dirs:

compile:
    [javac] Compiling 26 source files to C:\PX4\home\Firmware\Tools\jMAVSim\out\production\jMAVSim

create_run_jar:
      [jar] Building jar: C:\PX4\home\Firmware\Tools\jMAVSim\out\production\jmavsim_run.jar

copy_res:

BUILD SUCCESSFUL
Total time: 2 seconds
Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

(gdb)
(gdb) i args
No frame selected.
[... 59 lines cut ...]
 rip = 0x1005d101d in work_lpthread (../../src/platforms/common/work_queue/work_thread.c:296); saved rip = 0x0
 called by frame at 0x0
 source language c.
 Arglist at 0xff82c790, args: argc=<optimized out>, argv=<optimized out>
 Locals at 0xff82c790, Previous frame's sp is 0xff82c7c8
 Saved registers:
  rbx at 0xff82c7b8, rip at 0xff82c7c0, xmm15 at 0xff82c7c0
(gdb) where
#0  0x00000001005d101d in work_lpthread (argc=<optimized out>, argv=<optimized out>) at ../../src/platforms/common/work_queue/work_thread.c:296
#1  0x0000000000000000 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
(gdb) where
#0  0x00000001005d101d in work_lpthread (argc=<optimized out>, argv=<optimized out>) at ../../src/platforms/common/work_queue/work_thread.c:296
#1  0x0000000000000000 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
(gdb) where
#0  0x00000001005d101d in work_lpthread (argc=<optimized out>, argv=<optimized out>) at ../../src/platforms/common/work_queue/work_thread.c:296
#1  0x0000000000000000 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
(gdb) where
#0  0x00000001005d101d in work_lpthread (argc=<optimized out>, argv=<optimized out>) at ../../src/platforms/common/work_queue/work_thread.c:296
#1  0x0000000000000000 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
(gdb)
```

MaEtUgR, 2019-01-16:
It takes more time:
- check if it really works, I didn't incorporate it into the IDE and I'm so far unfamiliar with gdb command line usage
- add the debugger installation to the script
- generate new installer that includes it
- finally add it to the documentation

While a debugger could help with issues like https://github.com/PX4/Firmware/issues/11027 I think I'm fast by solving it printf style just now. That issue is right now really urgent for Windows use.

hamishwillee, 2019-01-16:
Not a problem. Let's just keep this open until you have time.

MaEtUgR, 2019-09-24:
GDB got into the toolchain thanks to @oneWayOut pushing it 👍 Sorry it took so long.
To use gdb out of the box you can use the Cygwin Toolchain >v0.6 linked as fast AWS download in the dev guide. Context: https://github.com/PX4/Firmware/pull/13016#issue-320767979

It would be nice to have documentation about how and configuration shortcuts to use it e.g. together with an IDE like VS code.

closed 2019-09-24 by MaEtUgR