PX4/PX4-windows-toolchain #11: Cannot find jinja2

SIGCHIAN, 2019-10-09:
Cannot find jinja2. pip does not work. I tried manually install jinja2 using pip but still get the same error.

CMake Error at D:/Program Files (x86)/CMake/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  couldn't find python module jinja2:

  for debian systems try:

        sudo apt-get install python-jinja2

  or for all other OSs/debian:

        sudo -H pip install jinja2

   (missing: PY_JINJA2)
Call Stack (most recent call first):
  D:/Program Files (x86)/CMake/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake/px4_find_python_module.cmake:61 (find_package_handle_standard_args)
  CMakeLists.txt:276 (px4_find_python_module)

-- Configuring incomplete, errors occurred!
See also "D:/PIXHAWKSource/Codes/Firmware/build/px4_sitl_default/CMakeFiles/CMakeOutput.log".
rm: cannot remove directory `/d/PIXHAWKSource/Codes/Firmware/build/px4_sitl_default': Permission denied
make: *** [px4_sitl] Error 1

hamishwillee, 2019-10-09:
This generally happens if you have multiple versions of python installed, and a package is installed in one pip but not the other. Any particular reason you aren't using the pre-built windows build?

MaEtUgR, 2019-10-16:
@SIGCHIAN What exactly is the context of that error. I'm assuming you try to build PX4 but you are not using the Cygwin Toolchain in this repository since the path `D:/Program Files (x86)/CMake/share/cmake-3.15/` tells me you're running a native installation of cmake without installing and/or running the cygwin environment that is supported with this toolchain. Here's the documentation: http://dev.px4.io/master/en/setup/dev_env_windows.html The toolchain that is supported and hosted through this repository is the "Cywin Toolchain".

Please provide more details in case I misunderstood or follow the guide if you try to use the Cygwin Toolchain.

MaEtUgR, 2020-02-19:
Since there was no follow-up to the suggestions and questions I'm closing this issue. If it's unsolved please feel free to reopen and provide further details.

closed 2020-02-19 by MaEtUgR