PX4/PX4-windows-toolchain #12: v0.8 having installation issue

wujoho, 2020-01-10:
My system specs:
A new installed Windows 10 Pro 64-bit in Oracle VM VirtualBox

Followed steps in https://dev.px4.io/master/en/setup/dev_env_windows_cygwin.html
and got the bellow errors at step 3 in the bash window:
[image]
[image]

Regards

simkslv, 2020-01-19:
I have the same problem on Win 7 pro and Win 10 pro. 
Did you install Visual studio at your win 10 on VM?

MaEtUgR, 2020-01-19:
Hey @wujoho @simkslv, thanks for the feedback. Apparently someone broke the simulation running on Windows. I presume it's a https://github.com/PX4/Firmware issue and not related to your toolchain installation. To verify you can check out the last stable release using the command `git checkout v1.10.0` and run simulation again `make posix jmavsim`.

> Did you install Visual studio at your win 10 on VM?

This toolchain has **NO** additional dependencies, there's no need to install Visual Studio.

wujoho, 2020-01-22:
Hi @MaEtUgR I still got error when using v1.10.0, attached my command lines:
[log.txt](https://github.com/PX4/windows-toolchain/files/4100582/log.txt)

It seems to be the OpenGL version issue in my VM?
[image]