PX4/PX4-windows-toolchain #13: Could NOT find PythonInterp: Found unsuitable version "2.7.14"

Katawann, 2020-01-20:
Hello,

I installed the windows toolchain, I fetch the repository and I try to build the simulation but the installation stop with this error:

`$ make px4_sitl jmavsim
-- PX4 version: v1.11.0-beta1-319-g44cac04abe
-- PX4 config file: /cygdrive/c/PX4/home/Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 7.3.0
-- The C compiler identification is GNU 7.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++.exe
-- Check for working CXX compiler: /usr/bin/c++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Could NOT find PythonInterp: Found unsuitable version "2.7.14", but required is at least "3" (found /usr/bin/python)
CMake Error at CMakeLists.txt:273 (message):
  Python 3 not found.  Please install Python 3:

      Ubuntu: sudo apt install python3 python3-devel python3-pip
      macOS: brew install python

-- Configuring incomplete, errors occurred!
See also "/cygdrive/c/PX4/home/Firmware/build/px4_sitl_default/CMakeFiles/CMakeOutput.log".
Error: /cygdrive/c/PX4/home/Firmware/build/px4_sitl_default is not a directory
make: *** [Makefile:202: px4_sitl] Error 1`

If I write `python --version` with the run-console.bat I have the output "python 2.7.14"

The same command with the shell give me "Python 3.8.1"...

Do I need to change something to allow the run-console.bat to find Python properly ? In the setup-environment.bat ?

VinniF, 2020-03-04:
Hi Katawann,

I had the same issue and discovered that it was due to an old link here (https://dev.px4.io/v1.9.0/en/setup/dev_env_windows_cygwin.html) which points to PX4 Cygwin Toolchain v0.5. Make sure that you install the most current version (v0.8) from here https://github.com/PX4/windows-toolchain/releases.

I suggest removing the old link pointing to an old version of the toolchain on S3 here https://dev.px4.io/v1.9.0/en/setup/dev_env_windows_cygwin.html.

hamishwillee, 2020-03-05:
The PX4 docs are versioned. That means that https://dev.px4.io/v1.9.0/en/setup/dev_env_windows_cygwin.html points to a version of the toolchain that will work with v1.9.

if you want to build for v1.11 (unreleased) you would look at the docs in master version of docs: https://dev.px4.io/master/en/setup/dev_env_windows_cygwin.html - they point to a version of the toolchain you can download that will work with the latest code.

> I suggest removing the old link pointing to an old version of the toolchain on S3 here https://dev.px4.io/v1.9.0/en/setup/dev_env_windows_cygwin.html.

As above, no :-). Use the correct docs for the version you want to build.

Katawann, 2020-03-05:
Hello both,

thank you for your answers! Forgot to update this issue. My error stops to show after I used the proper path for the folder (now C:\PX4 and before it was in my documents). Not sure if the run_console.bat could not find the proper Python version otherwise (maybe because of the setup_environnement.bat?).

I did not have the time to go further in debugging but it works after the PX4 folder has been moved to root C:

closed 2020-03-05 by Katawann