PX4/PX4-windows-toolchain #17: v0.7 additional steps before building

TheSoundMan000, 2020-03-29:
Installation of 0.7 failed to build without installing packaging and pyros-genmsg. Unsure if these are the expected versions so hoping to get this fixed up in the installer to remove any potential issues. Note that pip3 was deliberately not updated to keep in line with the default environment.

pip3 install packaging
pip3 install --user pyros-genmsg

The only thing to note is there was a prior installation of 0.5.

hamishwillee, 2020-03-29:
This is the Cygwin installation? 
Delete the PX4 folder and install the latest version - 0.9

TheSoundMan000, 2020-03-29:
Does 0.9 work with 1.10? Another user had indicated to only use the versions given in the documentation is https://dev.px4.io/v1.10/en/setup/dev_env_windows_cygwin.html which points to 0.7.

A fresh install on another PC has the same results unfortunately.

TSC21, 2020-03-29:
> Does 0.9 work with 1.10? Another user had indicated to only use the versions given in the documentation https://dev.px4.io/v1.10/en/setup/dev_env_windows_cygwin.html which points to 0.7. Perhaps a good idea to update the documentation if this is the case so newbies like myself don't get caught out.
> 
> A fresh install on another PC has the same results unfortunately.

Yes it does work with 1.10.

hamishwillee, 2020-03-29:
@TheSoundMan000 That's a good point. The answer being "Probably". IMO you'd be best of trying v0.8 because I am pretty sure that works with v1.10.
Otherwise you can wait until @MaEtUgR looks at this :-)

TheSoundMan000, 2020-03-29:
Awesome thanks guys, appreciate your help!

Happy to switch versions for now that's easy enough, just curious then is it safe to use the patch I put in place or does that leave the potential for incorrect versions etc?

hamishwillee, 2020-03-29:
I don't know what you mean by "patch you put in place". But my understanding is that everything lives in the PX4 folder - it is a complete environment. So if you delete/recreate that, the patch is gone.

MaEtUgR, 2020-03-30:
@TheSoundMan000 Sorry for any confusion. To work on latest PX4 master you can just use toolchain v0.9 but it might give you some small problems on PX4 v1.10.1 SITL I think because Cygwin updated the compiler. Your patch of adding the missing python packages is perfectly fine, that's exactly what the additional versions were triggered for. I added more release notes to hopefully clear things up: https://github.com/PX4/windows-toolchain/releases