PX4/PX4-windows-toolchain #18: Can't compile PX4 v1.9.2 under v0.9

bys1123, 2020-04-06:
Got error:
```
In file included from ../../src/lib/mixer/mixer_load.c:42:
../../src/lib/mixer/mixer_load.c: 在函数‘load_mixer_file’中:
../../src/lib/mixer/mixer_load.c:72:42: warning: Array subscript has type ‘char’ [-Wchar-subscripts]
   72 |   if ((strlen(line) < 2) || !isupper(line[0]) || (line[1] != ':')) {
      |                                      ~~~~^~~
```

bys1123, 2020-04-06:
I tried update arm gcc to `7-2018-q2-update` on ubuntu, but on ubuntu has no problem. And I run `make distclean` lots of times, still can't work.

bys1123, 2020-04-07:
I guess this is caused by in ubuntu gcc version is 7.5.0, but cygwin is 9.3.0
```
-- The CXX compiler identification is GNU 9.3.0
-- The C compiler identification is GNU 9.3.0
```

johan606303, 2020-04-21:
Hi I have news. I reached nothing with v1.10.2. But I think I have something with v1.11.0-beta1.

`git checkout v1.11.0-beta1`
`make px4_sitl_default none_iris`

then it will ask for updates 
[image]

I pressed `u`, then `git submodule update --init --recursive`
Then I pressed several times `u` untill

[image]
[image]
[image]

bys1123, 2020-04-21:
In my experience, run `git submodule update --init --recursive` first, then don't need to press `u` anymore.

johan606303, 2020-05-01:
It's not the case on my computer.
Does it work well for you ?

julianoes, 2020-05-04:
```
warning: Array subscript has type ‘char’ [-Wchar-subscripts]
```

This is a compiler warning. Can you try if you can compile it by disabling `-Werror` here:
https://github.com/PX4/Firmware/blob/960ad0693f3ff13c391bd8ddda61fd15dfe22ae0/cmake/px4_add_common_flags.cmake#L71

mdermksian, 2020-10-30:
> ```
> warning: Array subscript has type ‘char’ [-Wchar-subscripts]
> ```
> 
> This is a compiler warning. Can you try if you can compile it by disabling `-Werror` here:
> https://github.com/PX4/Firmware/blob/960ad0693f3ff13c391bd8ddda61fd15dfe22ae0/cmake/px4_add_common_flags.cmake#L71

This worked for me, but it's really odd - I feel like `line[0]` is very clearly using 0 to index into the array, which shouldn't be considered a char in any circumstance. Maybe my c/c++ knowledge is shaky...