PX4/PX4-windows-toolchain #19: Cant build Firmware, Need Help

Jsamsock, 2020-04-13:
I can not build firmware for PX4,  Im using v0.9 as given from this link: 
https://github.com/PX4/windows-toolchain/releases

I am getting the same problem as in a previous post:
https://github.com/PX4/Firmware/issues/9863

I tried some of the fixes mentioned above but have the same issue.  Can you help me figure this out?   I do mainly Matlab Programming and I do not have much experience with alot of this.  I attached a word document with the errors also.   
Thanks,
John

[image]

[image]

[error.docx](https://github.com/PX4/windows-toolchain/files/4471252/error.docx)

hamishwillee, 2020-04-13:
1. What are you building - master branch? 
2. Did you automatically get the Firmware branch as part of setting up 0.9 or did you do your own github clone?

Jsamsock, 2020-04-14:
> 
> 
>     1. What are you building - master branch?   
> 
>     2. Did you automatically get the Firmware branch as part of setting up 0.9 or did you do your own github clone?

I am not sure exactly what I am building.  I believe I automatically got the Firmware branch as part of setting up 0.9.   I have just been trying to follow the instructions given here for instance:

https://www.mathworks.com/help/releases/R2019a/supportpkg/px4/ug/px4-using-cygwin.html
and
https://dev.px4.io/v1.9.0/en/setup/dev_env_windows_cygwin.html

Thanks,
John

hamishwillee, 2020-04-15:
OK, so your build instruction is `make px4fmu-v5_default`. That instruction is pretty old - for current releases the build instruction is `make px4_fmu-v5_default`. I have tried the current instructions on this v0.9 cygwin setup and it works well.

What this tells me is that your version of PX4 is pretty old, and may not be compatible with the current cygwin environment. You're not building master for sure - I'd guess PX4v1.8 or maybe 1.9.

So you either need to revert to an old version of cygwin PX4 installer to match your build (e.g. v0.5) or update to the current master. 

This is not a defect and this is not the area for technical support. So I am closing. If you need further support please continue the discussion on the discussion forums: https://discuss.px4.io/c/software-development/13

closed 2020-04-15 by hamishwillee

Jsamsock, 2020-04-15:
> APL external email warning: Verify sender noreply@github.com before clicking links or attachments

Thanks for the help, I will give that a try.
John