PX4/PX4-windows-toolchain #21: How to use vscode IDE for build and debug?

lukegluke, 2020-08-29:
I've tried to use such script to run vscode
```
@ECHO OFF
PUSHD %~dp0
CALL toolchain\scripts\setup-environment.bat x
POPD

REM start logged in bash terminal
CALL bash -l -c "/cygdrive/c/PX4/vscode/Code.exe"

```
it sees cmake from toolchain, but then it goes mess with windows and cygwin pathes, for example
`PX4_SOURCE_DIR=c:/git/px4_firmware` while `PX4_BOARD_DIR=/cygdrive/c/git/px4_firmware/boards/px4/fmu-v5`
add error file `RELATIVE_PATH must be passed a full path to the directory` occured

dagar, 2020-08-29:
This isn't a great answer with respect to the PX4/windows-toolchain, but personally the best option I've found for PX4 development on Windows is to run native Windows VSCode and utilize one of the remote development plugins. https://code.visualstudio.com/docs/remote/remote-overview

VSCode remote WSL (or WSL2) works quite well other than running things like the gazebo gui (maybe that can be made to work now?). VSCode remote SSH to my Linux desktop also works very well, and I often use it by default when I'm away from my office.

lukegluke, 2020-08-29:
Thanks for suggestion! Yes, I saw support WSL, but didn't try it, it seemed that it wouldn't take "only takes a few minutes to setup" :)

dagar, 2020-08-30:
> Thanks for suggestion! Yes, I saw support WSL, but didn't try it, it seemed that it wouldn't take "only takes a few minutes to setup" :)

If you get Ubuntu LTS in WSL working the VSCode part will "just work". Native windows VSCode + remote WSL plugin and go.

lukegluke, 2020-08-30:
> If you get Ubuntu LTS in WSL working the VSCode part will "just work". Native windows VSCode + remote WSL plugin and go.

Ok, thanks, I will try. For total sure I suppose Ubuntu 18.04 LTS is your choice?

dagar, 2020-08-30:
> I suppose Ubuntu 18.04 LTS is your choice?

Both 20.04 and 18.04 should be fully supported right now.