PX4/PX4-windows-toolchain #22: Can't build the code on windows

OzanDurmaz, 2020-09-18:
Hi,

Trying to build the latest update PX4 v1.11 on windows. Followed the instructions, installed Windows Cygwin Toolchain v0.9 with github. When I try building the code for my cube black using “make px4_fmu-v3_default” I get the following screen massage:

ozand@DESKTOP-EV81A2B ~
$ cd firmware

ozand@DESKTOP-EV81A2B ~/firmware
$ make clean

ozand@DESKTOP-EV81A2B ~/firmware
$ make px4_fmu-v3_default
– PX4 version: v1.11.0-rc3-343-gc5f479dac7
– PX4 config file: /cygdrive/c/PX4/home/firmware/boards/px4/fmu-v3/default.cmake
– PX4 config: px4_fmu-v3_default
– PX4 platform: nuttx
– cmake build type: MinSizeRel
– The CXX compiler identification is GNU 7.3.1
– The C compiler identification is GNU 7.3.1
– The ASM compiler identification is GNU
– Found assembler: /cygdrive/c/PX4/toolchain/gcc-arm/bin/arm-none-eabi-gcc.exe
– Check for working CXX compiler: /cygdrive/c/PX4/toolchain/gcc-arm/bin/arm-none-eabi-g++.exe
– Check for working CXX compiler: /cygdrive/c/PX4/toolchain/gcc-arm/bin/arm-none-eabi-g++.exe – works
– Detecting CXX compiler ABI info
– Detecting CXX compiler ABI info - failed
– Detecting CXX compile features
– Detecting CXX compile features - failed
– Check for working C compiler: /cygdrive/c/PX4/toolchain/gcc-arm/bin/arm-none-eabi-gcc.exe
– Check for working C compiler: /cygdrive/c/PX4/toolchain/gcc-arm/bin/arm-none-eabi-gcc.exe – works
– Detecting C compiler ABI info
– Detecting C compiler ABI info - failed
– Detecting C compile features
– Detecting C compile features - failed
– Found PythonInterp: /usr/bin/python3 (found suitable version “3.6.9”, minimum required is “3”)
– build type is MinSizeRel
– PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-446-gfab242a
– Building and including px4_io-v2_default
– Found PythonInterp: /usr/bin/python3 (found version “3.6.9”)
– Using C++11
– Release build type: MinSizeRel
– ROMFS: px4fmu_common
– ROMFS: Adding rc.board_defaults
– ROMFS: Adding rc.board_sensors
– ROMFS: Adding rc.board_mavlink
– Found SVD:
– Configuring done
– Generating done
– Build files have been written to: /cygdrive/c/PX4/home/firmware/build/px4_fmu-v3_default
[0/1354] git submodule platforms/nuttx/NuttX/nuttx
[7/1354] git submodule platforms/nuttx/NuttX/apps
[10/1354] git submodule src/lib/ecl
[16/1354] git submodule mavlink/include/mavlink/v2.0
[18/1354] git submodule src/drivers/gps/devices
[19/1354] git submodule src/drivers/uavcan/libuavcan
[35/1354] Generating nuttx/binfmt/libbinfmt.a
FAILED: NuttX/nuttx/binfmt/libbinfmt.a
cd /cygdrive/c/PX4/home/firmware/build/px4_fmu-v3_default/NuttX/nuttx && find binfmt -type f -name *.o -delete && make -C binfmt --quiet --no-print-directory all TOPDIR=/cygdrive/c/PX4/home/firmware/build/px4_fmu-v3_default/NuttX/nuttx KERNEL=y EXTRAFLAGS=-D__KERNEL__
binfmt_globals.c:40:10: fatal error: nuttx/config.h: No such file or directory
#include <nuttx/config.h>
^~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:79: binfmt_globals.o] Error 1
[36/1354] Generating nuttx/libs/libxx/libxx.a
FAILED: NuttX/nuttx/libs/libxx/libxx.a
cd /cygdrive/c/PX4/home/firmware/build/px4_fmu-v3_default/NuttX/nuttx && find libs/libxx -type f -name *.o -delete && make -C libs/libxx --quiet --no-print-directory all TOPDIR=/cygdrive/c/PX4/home/firmware/build/px4_fmu-v3_default/NuttX/nuttx KERNEL=n EXTRAFLAGS=
libxx_cxa_atexit.cxx:40:10: fatal error: nuttx/config.h: No such file or directory
#include <nuttx/config.h>
^~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:84: libxx_cxa_atexit.o] Error 1
[38/1354] Generating nuttx/sched/libsched.a
FAILED: NuttX/nuttx/sched/libsched.a
cd /cygdrive/c/PX4/home/firmware/build/px4_fmu-v3_default/NuttX/nuttx && find sched -type f -name *.o -delete && make -C sched --quiet --no-print-directory all TOPDIR=/cygdrive/c/PX4/home/firmware/build/px4_fmu-v3_default/NuttX/nuttx KERNEL=y EXTRAFLAGS=-D__KERNEL__
clock/clock_initialize.c:41:10: fatal error: nuttx/config.h: No such file or directory
#include <nuttx/config.h>
^~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:57: clock_initialize.o] Error 1
[39/1354] Generating nuttx/mm/libmm.a
FAILED: NuttX/nuttx/mm/libmm.a
cd /cygdrive/c/PX4/home/firmware/build/px4_fmu-v3_default/NuttX/nuttx && find mm -type f -name *.o -delete && make -C mm --quiet --no-print-directory all TOPDIR=/cygdrive/c/PX4/home/firmware/build/px4_fmu-v3_default/NuttX/nuttx KERNEL=n EXTRAFLAGS=
mm_heap/mm_initialize.c:25:10: fatal error: nuttx/config.h: No such file or directory
#include <nuttx/config.h>
^~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:67: bin/mm_initialize.o] Error 1
[40/1354] Generating nuttx/fs/libfs.a
FAILED: NuttX/nuttx/fs/libfs.a
cd /cygdrive/c/PX4/home/firmware/build/px4_fmu-v3_default/NuttX/nuttx && find fs -type f -name *.o -delete && make -C fs --quiet --no-print-directory all TOPDIR=/cygdrive/c/PX4/home/firmware/build/px4_fmu-v3_default/NuttX/nuttx KERNEL=y EXTRAFLAGS=-D__KERNEL__
fs_initialize.c:40:10: fatal error: nuttx/config.h: No such file or directory
#include <nuttx/config.h>
^~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:94: fs_initialize.o] Error 1
[41/1354] Generating nuttx/boards/libboards.a
FAILED: NuttX/nuttx/boards/libboards.a
cd /cygdrive/c/PX4/home/firmware/build/px4_fmu-v3_default/NuttX/nuttx && find boards -type f -name *.o -delete && make -C boards --quiet --no-print-directory all TOPDIR=/cygdrive/c/PX4/home/firmware/build/px4_fmu-v3_default/NuttX/nuttx KERNEL=y EXTRAFLAGS=-D__KERNEL__
boardctl.c:25:10: fatal error: nuttx/config.h: No such file or directory
#include <nuttx/config.h>
^~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:94: boardctl.o] Error 1
[42/1354] Generating nuttx/libs/libc/libc.a
FAILED: NuttX/nuttx/libs/libc/libc.a
cd /cygdrive/c/PX4/home/firmware/build/px4_fmu-v3_default/NuttX/nuttx && find libs/libc -type f -name *.o -delete && make -C libs/libc --quiet --no-print-directory all TOPDIR=/cygdrive/c/PX4/home/firmware/build/px4_fmu-v3_default/NuttX/nuttx KERNEL=n EXTRAFLAGS=
assert/lib_assert.c:25:10: fatal error: nuttx/arch.h: No such file or directory
#include <nuttx/arch.h>
^~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:132: bin/lib_assert.o] Error 1
[43/1354] Generating nuttx/drivers/libdrivers.a
FAILED: NuttX/nuttx/drivers/libdrivers.a
cd /cygdrive/c/PX4/home/firmware/build/px4_fmu-v3_default/NuttX/nuttx && find drivers -type f -name *.o -delete && make -C drivers --quiet --no-print-directory all TOPDIR=/cygdrive/c/PX4/home/firmware/build/px4_fmu-v3_default/NuttX/nuttx KERNEL=y EXTRAFLAGS=-D__KERNEL__
bch/bchlib_setup.c:40:10: fatal error: nuttx/config.h: No such file or directory
#include <nuttx/config.h>
^~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:117: bchlib_setup.o] Error 1
[44/1354] Performing configure step for ‘px4io_firmware’
– PX4 version: v1.11.0-rc3-343-gc5f479dac7
– PX4 config file: /cygdrive/c/PX4/home/firmware/boards/px4/io-v2/default.cmake
– PX4 config: px4_io-v2_default
– PX4 platform: nuttx
– cmake build type: MinSizeRel
– The CXX compiler identification is GNU 7.3.1
– The C compiler identification is GNU 7.3.1
– The ASM compiler identification is GNU
– Found assembler: /cygdrive/c/PX4/toolchain/gcc-arm/bin/arm-none-eabi-gcc.exe
– Check for working CXX compiler: /cygdrive/c/PX4/toolchain/gcc-arm/bin/arm-none-eabi-g++.exe
– Check for working CXX compiler: /cygdrive/c/PX4/toolchain/gcc-arm/bin/arm-none-eabi-g++.exe – works
– Detecting CXX compiler ABI info
– Detecting CXX compiler ABI info - failed
– Detecting CXX compile features
– Detecting CXX compile features - failed
– Check for working C compiler: /cygdrive/c/PX4/toolchain/gcc-arm/bin/arm-none-eabi-gcc.exe
– Check for working C compiler: /cygdrive/c/PX4/toolchain/gcc-arm/bin/arm-none-eabi-gcc.exe – works
– Detecting C compiler ABI info
– Detecting C compiler ABI info - failed
– Detecting C compile features
– Detecting C compile features - failed
– Found PythonInterp: /usr/bin/python3 (found suitable version “3.6.9”, minimum required is “3”)
– build type is MinSizeRel
– PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-446-gfab242a
– Configuring done
– Generating done
– Build files have been written to: /cygdrive/c/PX4/home/firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware
ninja: build stopped: subcommand failed.
make: *** [Makefile:219: px4_fmu-v3_default] Error 1

ozand@DESKTOP-EV81A2B ~/firmware
$

Can someone help?

lukegluke, 2020-11-12:
It was fixed with:
https://github.com/PX4/PX4-Autopilot/pull/16040

MaEtUgR, 2021-02-24:
True, that one I fixed. Thanks for the hint to close the issue.

closed 2021-02-24 by MaEtUgR