PX4/PX4-windows-toolchain #24: make: *** No rule to make target `px4-v2'. Stop.

s-amal, 2021-02-20:
I have windows7 i downloded the ardupilot repository and px4 toolchaine, and when i build a target on px4 eclipse i get this +1

**** Build of configuration Default for project ArduCopter2 ****
make px4-v2
make: *** No rule to make target `px4-v2'. Stop.

Any ideas on what might be wrong ?

davids5, 2021-02-20:
the command is 
`make px4_fmv-v2`

MaEtUgR, 2021-02-24:
> i downloded the ardupilot repository
> Build of configuration Default for project ArduCopter2

@s-amal This toolchain is specifically tailored towards building the PX4 autopilot repository: https://github.com/PX4/PX4-Autopilot
I'm assuming you're confused between PX4 and Ardupilot. For full instructions on how to build PX4 please consult:
- https://docs.px4.io/master/en/dev_setup/dev_env_windows_cygwin.html
- https://docs.px4.io/master/en/dev_setup/building_px4.html

In summary, you should clone the PX4 repository and then run `make px4_fmv-v2` on the toolchains console.

closed 2021-02-24 by MaEtUgR

s-amal, 2021-03-04:
@MaEtUgR  no accually im not, im just trying to navigate the ardupilot repository , the px4 toolchaine eclipse im using it as an IDE as it says here    https://ardupilot.org/dev/docs/editing-the-code-with-eclipse.html