PX4/PX4-windows-toolchain #26: Failed to import numpy

Yecgaa1, 2021-03-20:
Sorry but I put up this issuse in this page
but please also help me to solve this problem
https://github.com/PX4/PX4-Autopilot/issues/17176
thanks again

Yecgaa1, 2021-03-23:
Numpy is a big question I think
```
❯ .\install-all-components.bat
* Automated PX4 Toolchain Components Installation
** Installing Cygwin
*** Downloading Cygwin Setup
****  Online  ****
  000000  ...
  14a013
CertUtil: -URLCache command completed successfully.
*** Installing Packages
Package list: cmake,gcc-g++,gdb,git,make,ninja,patch,xxd,nano,unzip,astyle,bash-completion,wget,libcurl-devel,procps-ng,moreutils,python27,python27-pip,python27-numpy,python27-jinja2,python27-tkinter,python3,python36-pip,python36-numpy,python36-jinja2,python36-tkinter
*** Cygwin Installation Done
** Installing Python Packages in Cygwin for PX4 Toolchain
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python2.7 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/cygdrive/c/Users/xtx/AppData/Local/Temp/pip-install-YOsdn6/pyulog/setup.py'"'"'; __file__='"'"'/cygdrive/c/Users/xtx/AppData/Local/Temp/pip-install-YOsdn6/pyulog/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /cygdrive/c/Users/xtx/AppData/Local/Temp/pip-pip-egg-info-tgPVRu
         cwd: /cygdrive/c/Users/xtx/AppData/Local/Temp/pip-install-YOsdn6/pyulog/
    Complete output (8 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/cygdrive/c/Users/xtx/AppData/Local/Temp/pip-install-YOsdn6/pyulog/setup.py", line 14, in <module>
        import versioneer
      File "versioneer.py", line 1739
        file=sys.stderr)
            ^
    SyntaxError: invalid syntax
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3.8.exe /usr/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /cygdrive/c/Users/xtx/AppData/Local/Temp/tmpqu4ivy39
         cwd: /cygdrive/c/Users/xtx/AppData/Local/Temp/pip-install-s_d1dd6p/numpy_3ec6c1528bb7406ba740a3a752c959dd
    Complete output (243 lines):
    Processing numpy/random/_bounded_integers.pxd.in
    Processing numpy/random/bit_generator.pyx
    Processing numpy/random/mtrand.pyx
    Processing numpy/random/_bounded_integers.pyx.in
    Processing numpy/random/_common.pyx
    Processing numpy/random/_generator.pyx
    Processing numpy/random/_mt19937.pyx
    Processing numpy/random/_pcg64.pyx
    Processing numpy/random/_philox.pyx
    Processing numpy/random/_sfc64.pyx
    Cythonizing sources
    blas_opt_info:
    blas_mkl_info:
    customize UnixCCompiler
      libraries mkl_rt not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/']
      NOT AVAILABLE

    blis_info:
      libraries blis not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/']
      NOT AVAILABLE

    openblas_info:
      libraries openblas not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/']
      NOT AVAILABLE

    atlas_3_10_blas_threads_info:
    Setting PTATLAS=ATLAS
      libraries tatlas not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/']
      NOT AVAILABLE

    atlas_3_10_blas_info:
      libraries satlas not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/']
      NOT AVAILABLE

    atlas_blas_threads_info:
    Setting PTATLAS=ATLAS
      libraries ptf77blas,ptcblas,atlas not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/']
[... 184 lines cut ...]
        egg_info.run()
      File "/cygdrive/c/Users/xtx/AppData/Local/Temp/pip-install-s_d1dd6p/numpy_3ec6c1528bb7406ba740a3a752c959dd/numpy/distutils/command/egg_info.py", line 24, in run
        self.run_command("build_src")
      File "/usr/lib/python3.8/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/cygdrive/c/Users/xtx/AppData/Local/Temp/pip-install-s_d1dd6p/numpy_3ec6c1528bb7406ba740a3a752c959dd/numpy/distutils/command/build_src.py", line 144, in run
        self.build_sources()
      File "/cygdrive/c/Users/xtx/AppData/Local/Temp/pip-install-s_d1dd6p/numpy_3ec6c1528bb7406ba740a3a752c959dd/numpy/distutils/command/build_src.py", line 161, in build_sources
        self.build_extension_sources(ext)
      File "/cygdrive/c/Users/xtx/AppData/Local/Temp/pip-install-s_d1dd6p/numpy_3ec6c1528bb7406ba740a3a752c959dd/numpy/distutils/command/build_src.py", line 318, in build_extension_sources
        sources = self.generate_sources(sources, ext)
      File "/cygdrive/c/Users/xtx/AppData/Local/Temp/pip-install-s_d1dd6p/numpy_3ec6c1528bb7406ba740a3a752c959dd/numpy/distutils/command/build_src.py", line 378, in generate_sources
        source = func(extension, build_dir)
      File "numpy/core/setup.py", line 444, in generate_config_h
        moredefs, ignored = cocache.check_types(config_cmd, ext, build_dir)
      File "numpy/core/setup.py", line 49, in check_types
        out = check_types(*a, **kw)
[... cut ...]
WARNING: Discarding https://files.pythonhosted.org/packages/d2/48/f445be426ccd9b2fb64155ac6730c7212358882e589cd3717477d739d9ff/numpy-1.20.1.zip#sha256=3bc63486a870294683980d76ec1e3efc786295ae00128f9ea38e2c6e74d5a60a (from https://pypi.org/simple/numpy/) (requires-python:>=3.7). Command errored out with exit status 1: /usr/bin/python3.8.exe /usr/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /cygdrive/c/Users/xtx/AppData/Local/Temp/tmpqu4ivy39 Check the logs for full command output.
```

hamishwillee, 2021-03-24:
@MaEtUgR One for you. 

@Yecgaa1 If you're blocked on this, you may be able to work using a virtual machine http://docs.px4.io/master/en/dev_setup/dev_env_windows_vm.html

Yecgaa1, 2023-10-12:
I'm so sorry this issuse continued for a long time. Yeah, use virture mechine can solve this problem.

> @MaEtUgR一份给你。
> 
> @Yecgaa1如果您被阻止，您也许可以使用虚拟机http://docs.px4.io/master/en/dev_setup/dev_env_windows_vm.html

closed 2023-10-12 by Yecgaa1