PX4/PX4-windows-toolchain #28: New v1.12.3 build breaks because CMake version is outdated

jatast, 2021-12-03:
The latest version of PX4 v1.12.3 breaks the current toolchain due to the automatically installed version of cmake being too old (current version 3.14.5, requires 3.16+). I'm using Windows toolchain version: v0.9. Can this get an update?

Output:
CMake Error at src/drivers/px4io/CMakeLists.txt:55 (cmake_policy):
  Policy "CMP0097" is not known to this version of CMake.

Line causing problem:
cmake_policy(SET CMP0097 NEW) 
In file: src/drivers/px4io/CMakeLists.txt on line 55

[Cmake documentation](https://cmake.org/cmake/help/git-stage/policy/CMP0097.html)

hamishwillee, 2021-12-05:
@MaEtUgR Is this something we could patch easily? We're getting pretty close to the point where maybe a WSL solution could reasonably be used instead

MaEtUgR, 2022-01-31:
I know there is an older CMake version in the currently release toolchain and I would gladly update it if this issue wouldn't prevent me from generating a new installer: https://github.com/PX4/PX4-windows-toolchain/issues/31

hamishwillee, 2022-02-02:
FYI This works for master and `px4_fmu-v5` with the default 0.9 installer (didn't try v1.12.3). Build ended with:
```
Creating /cygdrive/c/PX4/home/Firmware/build/px4_fmu-v5_default/px4_fmu-v5_default.px4
```

Is there a manual patch? I.e. can I run some kind of installer script from within the console to update cmake? If so we could update the docs until this is sorted. Otherwise I guess we add a note "does not work for now"?

jatast, 2022-02-10:
> @MaEtUgR Is this something we could patch easily? We're getting pretty close to the point where maybe a WSL solution could reasonably be used instead

Issue with WSL is the latest version of WSL2 still does not play nicely with windows com ports. You'd be forced to use WSL 1 if you want to upload to FC.

hamishwillee, 2022-02-10:
@jatast We're aware. You can separately use QGC on Windows for that, if needed, since it can see the WSL file system.