PX4/PX4-windows-toolchain #30: Missing packages after install

chrismiller7, 2022-01-21:
After installing v0.8 or v0.9, running "make px4_sitl jmavsim" causes several errors with missing packages. On a fresh install, I have to manually add the packages:

run run-console.bat
cd Firmware

pip3 install --upgrade pip
pip3 install kconfiglib
pip3 install --user jsonschema
pip3 install future

make px4_sitl jmavsim

hamishwillee, 2022-01-30:
Thanks @chrismiller7 

@MaEtUgR Howdy! What's the best way to sort this. We should add these dependencies, but are we closer to an updated WSL2 solution that can use the linux PX4 install scripts instead?

MaEtUgR, 2022-01-31:
@chrismiller7 Thanks a lot for reporting this! I would change the toolchain immediately but the biggest problem I currently have with the Cygwin toolchain is that the Cygwin base system grew significantly in size and is now bigger than what fits in a single file MSI installer. So whenever I try to update the toolchain and build new versions I cannot create an installer for it anymore. This would need some kind of file splitting in the installer generation which I'm not yet familiar with. We could at least add the necessary commands to the Cygwin toolchain documentation to enable users in the meantime.

@hamishwillee Yes, I finished writing the basic instructions: https://github.com/PX4/PX4-user_guide/pull/1747 The instructions use 1:1 the existing setup scripts. If you're interested @chrismiller7 you can give this a try. I'm happy to hear any feedback.

hamishwillee, 2022-02-02:
OK, I have created a patch for the cygwin docs in https://github.com/PX4/PX4-user_guide/pull/1755. The note can be removed when this issue is fixed.