PX4/PX4-windows-toolchain #34: install-cygwin-python-packages.bat errors out on Windows 10

copterspace, 2022-07-15:
I try to install latest v1.0 toolchain on Windows 10
It does not have installer. I download source code to C:\PX4 and run install-all-components.bat

It errors out on install-cygwin-python-packages.bat:
```
** Installing Python Packages in Cygwin for PX4 Toolchain
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python2.7 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/cygdrive/c/Users/IPETRO~1.BYT/AppData/Local/Temp/pip-install-VdtpnI/pyulog/setup.py'"'"'; __file__='"'"'/cygdrive/c/Users/IPETRO~1.BYT/AppData/Local/Temp/pip-install-VdtpnI/pyulog/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base '/cygdrive/c/Users/IPETRO~1.BYT/AppData/Local/Temp/pip-pip-egg-info-uMBNA8'
         cwd: /cygdrive/c/Users/IPETRO~1.BYT/AppData/Local/Temp/pip-install-VdtpnI/pyulog/
    Complete output (8 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/cygdrive/c/Users/IPETRO~1.BYT/AppData/Local/Temp/pip-install-VdtpnI/pyulog/setup.py", line 14, in <module>
        import versioneer
      File "versioneer.py", line 1739
        file=sys.stderr)
            ^
    SyntaxError: invalid syntax
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```
How to install latest toolchain on Windows 10 properly? Is it supposed to be used on Windows10?

hamishwillee, 2022-07-17:
@MaEtUgR Suggestions?

@copterspace FYI personally I use http://docs.px4.io/main/en/dev_setup/dev_env_windows_wsl.html#windows-wsl2-based-development-environment because it works for building the head revision of PX4, and I am not sure that Cygwin does at the momement (Matthias, can you confirm)?

For Windows 10 you have to run the simulations HEADLESS.

copterspace, 2022-07-18:
Thanks for suggestion. I use Jmavsim, so headless run is not sufficient for my case.