PX4/PX4-windows-toolchain #36: make px4_sitl jmavsim failed

scps990210, 2022-09-17:
Excuse me, i installed px4 by Amazon S3 and do the instructs in [this](https://dev.px4.io/v1.11_noredirect/en/setup/dev_env_windows_cygwin.html#getting_started) step by step. But when i do the command make px4_sitl jmavsim it encountered some problems below 
[image]

hamishwillee, 2022-09-18:
That should work if you are targeting PX4 v1.11; perhaps @MaEtUgR can comment.

If it was me I would use WSL2 - following the instructions in http://docs.px4.io/main/en/dev_setup/dev_env_windows_wsl.html but using the PX4 v1.11 sources.

scps990210, 2022-09-19:
Thanks for your reply! I have followed the steps in http://docs.px4.io/main/en/dev_setup/dev_env_windows_wsl.html to install wsl2 but when i type "make px4_sitl " it show "ninja: no work to do.".
and when i type "make px4_sitl jmavsim" it still encounter some problems below 
[image]
i think i have problems about how to set the version to 1.11 while doing the work in http://docs.px4.io/main/en/dev_setup/dev_env_windows_wsl.html

hamishwillee, 2022-09-19:
Any reason why you have to stay at PX4 v1.11? I'll try look at this on Wednesday.

scps990210, 2022-09-19:
OK, thank you! 
Because you suggest me to follow the instructions in http://docs.px4.io/main/en/dev_setup/dev_env_windows_wsl.html but using the PX4 v1.11 sources. But i don't know how to use the PX4 v1.11 sources.

Quietly-20201113, 2024-03-05:
I built it in window11 and followed the tutorial in https://dev.px4.io/v1.11_noredirect/zh/setup/dev_env_windows_cygwin.html, but I received the following error message. I can't upgrade cmake anyway. Please tell me. What should be done
```html
-- stdout output is:
loading initial cache file /cygdrive/c/PX4/home/Firmware/build/px4_sitl_default/src/modules/uxrce_dds_client/tmp/libmicroxrceddsclient_project-cache-RelWithDebInfo.cmake
-- Configuring incomplete, errors occurred!

-- stderr output is:
CMake Error at CMakeLists.txt:18 (cmake_minimum_required):
  CMake 3.16 or higher is required.  You are running version 3.14.5

CMake Error at /cygdrive/c/PX4/home/Firmware/build/px4_sitl_default/src/modules/uxrce_dds_client/src/libmicroxrceddsclient_project-stamp/libmicroxrceddsclient_project-configure-RelWithDebInfo.cmake:47 (message):
  Stopping after outputting logs.

[19/235] Building CXX object src/systemcmds/tests/CMakeFiles/systemcmds__tests.dir/test_matrix.cpp.o
ninja: build stopped: subcommand failed.
make: *** [Makefile:232：px4_sitl] 错误 1
```

hamishwillee, 2024-03-05:
We're up to PX4 v1.14 - I suggest you migrate up from PX4 v1.11 and you're more likely to get support. 

If I had to get this working I would TRY to do this on WSL2 - so follow the WSL2 setup instructions from https://docs.px4.io/main/en/dev_setup/dev_env_windows_wsl.html#overview BUT install Ubuntu 20.04 rather than 20.02

Then use https://github.com/PX4/PX4-Autopilot/blob/v1.11.0/Tools/setup/ubuntu.sh as per the instructions to install the development environment.

And fetch PX4 v1.11 release branch and build that. 

Note also that I would try Gazebo rather than jmavsim, which is now "community supported">

I'm closing this, as the Cygwin windows toolchain is no longer supported.

closed 2024-03-05 by hamishwillee